PeteLawler/BeagleBrew #50: Merging / adding RasPi compatibility

guysoft, 2017-10-01:
Hey,
So I want to make a RaspberryPi distro that would have RasPiBrew. And I saw this repo did a lot of good progress, so I was thinking of using it as the version to ship. However its been adapted to the BeagleBoard, and I was wondering - how much work is it going to be to add an option to switch between boards and support both?

I think I can write two modules with a common functions for on and off. One will use Adafruit_BBIO and other the Raspberrypi GPIO library.
Then a config variable decides which one should be imported and used.

Is this something you are willing to accommodate if I pull request?

(Also can't see any issue tracker on gitlab)

PeteLawler, 2017-10-02:
Originally I used a third setting on the GPIO config for differentiating between RPi and BBB. The 'main' thing, I guess, is that the Adafruit BBIO library uses a string for the GPIO pin name and the RPi library uses a number.

To be honest, on the 'stable' branch there's not much that's changed from the RasPiBrew except for as noted above and shoving it all through pylint and fixing it up there.

I have been contemplating 'bringing back' the RasPi as a supported platform, which will require me to go out and buy one which shouldn't be too far away really.

I'll look into the issue tracker on GitLab. More than happy to look at requests etc. Apologies for the inconvenience, and thanks for taking the time to contact me :)

guysoft, 2017-10-02:
Awesome. It should be easy for me to add and test support for the Pi. Will just need you to confirm I didn't break anything on the beagle side. If you open an issue tracker on gitlab, we can take the conversation there.
As long as we are both talking we could figure it out.

PeteLawler, 2017-10-02:
Please confirm that you can now see the GL issue tracker as requested.

guysoft, 2017-10-02:
Confirm, yay both are issue #50https://gitlab.com/petelawler/BeagleBrew/issues/50